farrellf/TelemetryViewer #3: is it possible to read the data from the socket ?

aveshd, 2017-07-08:
Hi,

is it possible to read the data from the socket ? In my usage I have data coming through UDP, it could be in the same format. 

If you don't have this, could you please point me to the area of the code which would need modification to send the data for processing. 

Thank you

fasse-sweden, 2017-07-22:
I also would like a way to feed the data wireless. Udp or through mqtt. I too could maybe help with some code.

farrellf, 2017-07-26:
I plan to add support for TCP and UDP in the next version, but that is probably a few months away as I have little spare time these days.

Version 0.4 involved refactoring some of the code in preparation for TCP and UDP support. The packet processing code now reads from InputStreams, so it should be trivial to wedge in support for various protocols.

If you feel like modifying the code to suit your needs, the related code should just be the connectToSerialPort() function in Controller.java, and the ControlsPanel.java class that handles the GUI side of things:

https://github.com/farrellf/TelemetryViewer/blob/master/Telemetry%20Viewer/src/Controller.java?ts=4#L323

https://github.com/farrellf/TelemetryViewer/blob/master/Telemetry%20Viewer/src/ControlsRegion.java?ts=4

farrellf, 2018-04-05:
Just an update: I have written the TCP and UDP code, but I'm working on making it more efficient and robust. I've been busy with work. Hope to have things cleaned up enough for another release within a month or two.

MiloMindbender, 2018-04-06:
Is there a version in github that does this that I could test?

MiloMindbender, 2018-05-17:
Hi farrellf 
If you are working on TCP support I would like to help you develop and test it.  I am also using it for a self balancing robot that sends telemetry by TCP text stream.  Right now I use a virtual com port driver that connects the tcp stream to com13 so your software can see it.  Unfortunately the license for the virtual com port is expensive so I would really like to help you get your TCP support working!

Wooopz, 2019-02-02:
After much research, I was able to connect the telemetry viewer via a wifi connection using the Windows "Virtual Serial Port" driver and the "Esp Link" firmware in esp8266.
You will find useful information here:
https://caron.ws/diy-cartes-microcontroleurs/raspberrypi/connexion-esp8266/

aveshd, 2019-02-09:
Thank you for implementing this. You can close the issue, will try the newer version and let you know if there are any issues.

closed 2019-02-10 by farrellf