farrellf/TelemetryViewer #4: Option for constant data range on y-axis instead of auto ranging

stevekramer99, 2017-07-18:
Hi, This is really great, perfect for robot development!  

I would REALLY like to see an option for constant y-axis range.  This is really important because when you are across the room, it would be really good to know where the values are in the range that you specify without having to look at the axis values, you just know from the position of the line.  Thanks!

P.S.  Could you add horizontal bar charts for time data so you could more easily compare several values visually at the same time.  ONE more suggestion - some simple signal processing options such as averaging, LPF, HPF - so the arduino doesn't have to  :)

jcrodriguez40, 2017-07-18:
Hey Steve. 

The first feature you ask for is already implemented. Whenever you ad a chart, look down where it says "Y-axis Maximum/Minimum" (in the case of a Dial chart, it'll say "Dial Maximum/Minimum"). By editing this field (unticking the "Automatic" checkbox), you get a fixed-range Y-axis. 

Hope it helps!

stevekramer99, 2017-07-18:
Hey I just found v0.3, I was using 0.2.  I initially found it through YouTube BTW, but the earlier vids don't reference the now updated versions.  Thanks for maintaining this.  I hope it gets featured in some major Arduino propaganda.

jcrodriguez40, 2017-07-18:
Actually, the exact same thing happened to me too. I've been using the v0.2 release for a couple of months and only last week I found out that there was a v0.3 release. 

I am also in debt with the creator of this app. My work has been inmensely impacted by the quality of the visuals of the TelemetryViewer. 

BTW, I do not use arduino but STM32-based uC. It still works like a charm! My point is ... I hope this tool gets the recognition it deserves, no matter where! :D

farrellf, 2017-07-26:
I have updated the YouTube video description to point to my website now, instead of specific Jar files. That should make it easier for people who run across the older videos. Here's a link:
http://www.farrellf.com/TelemetryViewer/

I plan to add support for computed datasets, but it might be a while before I get to that. LPF, HPF, BPF, mean, median, math functions, etc. Basically just like an oscilloscope.

I also prefer the STM32 microcontrollers. I don't actually use Arduinos very much but most of my YouTube viewers use them so I try to aim the videos at them. That balancing robot uses an STM32F051, and I use the F4 series for more advanced projects.