farrellf/TelemetryViewer #5: serial send

jschoch, 2017-10-21:
would be nice to be able to send since you can't send with anything else without disconnecting.

jcrodriguez40, 2017-10-22:
True. 

A workaround might be using 2 serial ports. Many microcontrollers come with several HW serial ports built in. However, even in the case of "poorly-equipped" chips (like some of ATMEL's), you might get away with it by using the "Software_Serial" Library. 

Hope it helps!

farrellf, 2018-04-05:
I plan to add support for this, but it might be a while before I get to it.

farrellf, 2021-07-25:
I finally got around to implementing this. It's in the latest release:

http://www.farrellf.com/projects/software/2021-07-24_Telemetry_Viewer_v0.8/
https://www.youtube.com/watch?v=NUBm9b8YQMQ&t=937s

closed 2021-08-18 by farrellf

jschoch, 2021-08-20:
sweet, thanks!