farrellf/TelemetryViewer #6: Java errors

andresrobinson, 2017-10-29:
Hi, I tried multiple Java versions, both 32 and 64 bit and I'm having problems with all of them.
With java version JRE-9.0.1:
```C:\Users\andre>"C:\Program Files\Java\jre-9.0.1\bin\java.exe" -jar C:\Users\andre\Desktop\TelemetryViewer_v0.3.jar
Exception in thread "main" java.lang.reflect.InvocationTargetException
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
        at java.base/java.lang.reflect.Method.invoke(Unknown Source)
        at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:58)
Caused by: java.lang.NoClassDefFoundError: com/jogamp/opengl/GLCapabilitiesImmutable
        at Main.main(Main.java:16)
        ... 5 more
Caused by: java.lang.ClassNotFoundException: com.jogamp.opengl.GLCapabilitiesImmutable
        at java.base/java.net.URLClassLoader.findClass(Unknown Source)
        at java.base/java.lang.ClassLoader.loadClass(Unknown Source)
        at java.base/java.lang.ClassLoader.loadClass(Unknown Source)
        ... 6 more
```

With JRE 1.8.0_151

```
2017-10-29 14:47:52.904:INFO::main: Logging initialized @6553ms to org.eclipse.jetty.util.log.StdErrLog
Exception in thread "main-AWTAnimator#00" java.lang.NullPointerException
        at LogitechSmoothScrolling.enableSmoothScrolling(LogitechSmoothScrolling.java:101)
        at Main$1.windowGainedFocus(Main.java:35)
        at java.awt.Window.processWindowFocusEvent(Unknown Source)
        at java.awt.Window.processEvent(Unknown Source)
        at java.awt.Component.dispatchEventImpl(Unknown Source)
        at java.awt.Container.dispatchEventImpl(Unknown Source)
        at java.awt.Window.dispatchEventImpl(Unknown Source)
        at java.awt.Component.dispatchEvent(Unknown Source)
        at java.awt.KeyboardFocusManager.redispatchEvent(Unknown Source)
        at java.awt.DefaultKeyboardFocusManager.typeAheadAssertions(Unknown Source)
        at java.awt.DefaultKeyboardFocusManager.dispatchEvent(Unknown Source)
        at java.awt.Component.dispatchEventImpl(Unknown Source)
        at java.awt.Container.dispatchEventImpl(Unknown Source)
        at java.awt.Window.dispatchEventImpl(Unknown Source)
        at java.awt.Component.dispatchEvent(Unknown Source)
        at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
        at java.awt.EventQueue.access$500(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue$4.run(Unknown Source)
        at java.awt.EventQueue$4.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue.dispatchEvent(Unknown Source)
        at java.awt.SequencedEvent.dispatch(Unknown Source)
        at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
        at java.awt.EventQueue.access$500(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue$4.run(Unknown Source)
        at java.awt.EventQueue$4.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue.dispatchEvent(Unknown Source)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.run(Unknown Source)
[... 66 lines cut ...]
        at java.awt.EventQueue$4.run(Unknown Source)
        at java.awt.EventQueue$4.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue.dispatchEvent(Unknown Source)
        at java.awt.SequencedEvent.dispatch(Unknown Source)
        at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
        at java.awt.EventQueue.access$500(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue$4.run(Unknown Source)
        at java.awt.EventQueue$4.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue.dispatchEvent(Unknown Source)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.run(Unknown Source)
```

farrellf, 2018-04-05:
It should work with Java 8. The serial port and OpenGL libraries I'm using seem to have issues with Java 9. I'm looking into workarounds.

The error about "LogitechSmoothScrolling.java" should be harmless, but I will fix my code to prevent that.

It looks like the main problem is "com.jogamp.opengl.GLException: Method "glGenFramebuffers" not available" What OS and video card are you using? Are you using the latest graphics drivers?

j-opificius, 2018-05-17:
If you don't mind I'll give my answer to those questions, as I'm experiencing the same Logitec error...
Windows 10 Pro on a Dell laptop, 
Video is reported as Intel HD Graphics 530, with NVIDIA Quadro M 1000M.
I will also say I have miserable luck with Java 10: the .jar takes minutes to open up, and when it does the display area array (8 x 8) is sized for the original resolution of the screen, not the current display resolution  (I'm currently using 1920 x 1080, so as to be more consistent with my external monitor of that resolution).
I resolved the startup delay by uninstalling JDK10, installing JDK8, and shifting the "Javapath" entries in the System Path environment variable to after my manually added references to Java 8. Fortunately that also fixed the incorrect screen resolution.

farrellf, 2020-07-19:
I just released a new version, and it now works in Java 9+. I also upgraded to a pre-release version of the OpenGL library that I use, and rewrote most of my OpenGL code. It should now work on anything that supports OpenGL 3.2 Core Profile, or OpenGL ES 3.1 with the GL_EXT_geometry_shader extension.

Please try it out and let me know if you still have problems:

http://farrellf.com/projects/software/2020-07-17_Telemetry_Viewer_v0.7/

closed 2020-09-04 by farrellf