farrellf/TelemetryViewer #7: Required version of JRE

Defragster, 2018-01-16:
I didn't see it noted anywhere what version of JRE works with TelemetryViewer.
Installing JRE 9x64 on my Win10x64 machine would not respond to opening the JAR.

Uninstalled JRE 9 and installed JRE 8x64 and it came right up.

Looking forward to working with it, it looks like an impressive piece of work!
It would be nice for Arduino IDE like behavior is there was a terminal window where non data messages ( debug text ) was displayed.  Also as I saw as already an issue - a write terminal window so the device under observation could be passed outgoing messages to alter behavior on command.

farrellf, 2019-02-15:
Right now it requires Java 8. I keep finding more issues when testing in Java 9/10/11. The show stopper right now is that the OpenGL library I use does not work well with high-DPI displays in Java 11. The GLCanvas is the wrong size, etc. And sadly, it looks like that library is barely maintained anymore, so it might be time to find an alternative.

farrellf, 2020-07-19:
I just released a new version, and it now works in Java 9+.

http://farrellf.com/projects/software/2020-07-17_Telemetry_Viewer_v0.7/

closed 2020-07-19 by farrellf