farrellf/TelemetryViewer #8: Export Time?

NeuroKnight, 2018-02-02:
Hi Farrell,

On the exporting CSV, is it possible to also export the time stamp or sample count in each line? I just want it as another variable in the line.

Thank you!

farrellf, 2018-04-05:
I plan to add support for that soon. Right now there is no logging of when samples were received.

NeuroKnight, 2018-04-05:
Awesome! I look forward to the next update. It would be great to have a time bar for the x axis and for this to be an exportable column on the csv. Cheers!

Alasterer, 2018-10-24:
This function would also be extremely helpful with my use case.
Looking forward to it!

EndUser12, 2018-11-28:
Hello Farrell,

Has there been any updates on this? We would really like to have a timestamp on our data.

Thank you!

farrellf, 2018-12-06:
I still plan to implement this, probably in time for the next release. But I have very little free time these days, so I don't know how soon that will be. Maybe several months from now.

EndUser12, 2018-12-13:
> I still plan to implement this, probably in time for the next release. But I have very little free time these days, so I don't know how soon that will be. Maybe several months from now.

Great! Thanks for the update.

NeuroKnight, 2018-12-14:
> > I still plan to implement this, probably in time for the next release. But I have very little free time these days, so I don't know how soon that will be. Maybe several months from now.
> 
> Great! Thanks for the update.

Yes, thank you. Good luck!

farrellf, 2019-09-18:
This issue is solved with version 0.6. Timestamps are now included in the CSV logs, and you can even import (replay) the logs.

Relevant commits are:
https://github.com/farrellf/TelemetryViewer/commit/26ae7538c275213b58284e9dba1e063485f21d82
https://github.com/farrellf/TelemetryViewer/commit/b7ec0be48fa7d204c6e1ea08f5eb757fcbd3adc4
https://github.com/farrellf/TelemetryViewer/commit/450d90782a5a200ecabc151759d5571fa91cc7f7

Please try v0.6, and if things seem fine, close this issue.
http://farrellf.com/TelemetryViewer/TelemetryViewer_v0.6.jar

closed 2020-06-12 by farrellf