farrellf/TelemetryViewer #9: Serial over Bluetooth (Rfcomm)

vsaljooghi, 2018-02-11:
Hello

Big thanks for the great job you have done. 
I want to plot data that comes through Bluetooth from Arduino. I am able to receive data on /dev/rfcomm0. But unfortunately, this device node will not be enumerated in Telemetry GUI. I went through the source code and saw that the following function is responsible for it: SerialPort.getCommPorts().

Could please give me some hint on how to solve this issue?

Best Regards,

farrellf, 2018-04-05:
I use an HC-06 Bluetooth module in Windows 10 with no problems. I'm guessing this might be an issue with the serial port library I use. Could you try the same Bluetooth module with a Windows computer to see if the problem only occurs in Linux?

farrellf, 2019-02-02:
Unable to duplicate this problem, and no response after months, so I'm closing this issue.

closed 2019-02-02 by farrellf