farrellf/TelemetryViewer #10: "Save Layout" function is not working in Binary Mode.

Wooopz, 2018-02-16:
The function "Save Layout" is not working in Binary Mode (it works in CSV but not in Binary).
When you click on "Save Layout" button in binary mode, a empty file is generated.
It is really boring to reconfigure all layout each time i restart the software.
Could you correct this bug please. Thank you.

Lecerof, 2018-03-24:
Hi, do you mind testing out the version I have on my github. I think I have resolved this issue but I am not sure because I do not have any good means of testing it out. Athough, the layout is slightly different on my version, I think you will understand it without any further instructions.

farrellf, 2018-04-05:
It looks like Lecerof fixed the problem by checking if the ChecksumProcessor is null. I've always used a checksum so forgot about that. Wooopz, can you confirm if that fixed your problem?

Wooopz, 2018-04-06:
Yes. The Lecerof version fixed the problem. Thank you.

farrellf, 2019-02-02:
This was fixed as part of version 0.5:
https://github.com/farrellf/TelemetryViewer/commit/3f37052542cb0d5df20aa85120ef81fa4a443fb9

If there are no objections, I'll close this issues in a few days.

closed 2019-02-15 by farrellf