farrellf/TelemetryViewer #11: Mac Issue

BcGh01, 2018-02-20:
Great software ! Just a couple of issues.

I am running on a Mac with macOS - High Sierra 10.13.3. The widget controls in the top right of graphs do not work. I cannot close or change the setting. 

Also have just upgraded to Java 9.0.4 and the software will not run anymore. Message reads "The Java JAR File "TelemeteryViewer_v0,4 could not be launched."

Help with these two issues would be great. Thank you

Lecerof, 2018-03-22:
I think it might be due to the retina display and the different way coordinates work with windows and mac, where the retina macbook uses points (i.e 2x2 pixels) rather than absolute pixel values. I am currently searching for a solution to this problem myself. I can notify you if I manage to solve it.

BcGh01, 2018-03-22:
Great to hear from you and thank you for your reply. 

Yes, the widget size is very small to there may not be enough area to detect the click.

I am very happy to test any solution you have and if you send me instructions on how to compile the code I would be happy to help.

Also the “Statistics” options, shown in the red box below, do not appear on the Mac either 

Thanks again and if I can help please let me know.
Brad

Lecerof, 2018-03-22:
Any solution I come up with will end up on my github since I am not a collaborator on this project, it is easy to find it there. In what environment do you use it/compile it?

BcGh01, 2018-03-22:
I am using a MacBook Pro running High Serra 10.13.3. Screen 15.4in ( 2880 X 1800 )

Lecerof, 2018-03-22:
Try to pull down my version from my github now. Dont know how to help you with the compilation though, do you compile it in the terminal and then run it through the terminal aswell?

BcGh01, 2018-03-22:
Thank you. I have your version and will work on it. I will send an update soon.

Lecerof, 2018-03-23:
Does it work?

BcGh01, 2018-04-24:
Still have not managed to generate a JAR file. Not sure how to compile your code. Can you help please

BcGh01, 2018-04-24:
> I usually compile it through Eclipse. What environment do you use?

I have Eclipse but only used it for Android development . Not sure of the setup I need for a JAR output
Could you let me know how you set things up please

Lecerof, 2018-04-24:
I setup a new project through git in eclipse. Its really easy: https://stackoverflow.com/questions/9192125/how-to-create-new-project-from-git-repo-using-eclipse-egit?utm_medium=organic&utm_source=google_rich_qa&utm_campaign=google_rich_qa

BcGh01, 2018-04-24:
THANK YOU ! ! 
I will look at this today

farrellf, 2020-07-19:
While trying to get Telemetry Viewer to work in Java 9+ I encountered similar problems. I just released a new version, and I suspect it may have solved this problem. (I don't have access to a Mac, so I have no way to verify it.)

http://farrellf.com/projects/software/2020-07-17_Telemetry_Viewer_v0.7/

closed 2020-09-04 by farrellf