farrellf/TelemetryViewer #12: Question on max input size

mjs513, 2018-04-01:
Hi
First this is a great viewer for telemetry. So this is more of a question than an issue, well maybe.

We are probably pushing this to the limits but we are passing it 36 floats for a total length of 1116 chars on input to the viewer.  I can't seem to get it to display anything.  Was wondering if its me or there is some sort of max size on the number of input chars?

Thanks
Mike

Lecerof, 2018-04-05:
Does it work with less floats?

mjs513, 2018-04-05:
Hi Lecerof,
Finally got it working.  Was a problem with data format.  When I posted it looked ok but didn't work.  I reformatted the data and now have it working.  Up to 77 floats being sent.  But with that many I can't get all them on the chart window.  Really needs to be a scrolling window.  Going to close as that's a separate issue

Mike

closed 2018-04-05 by mjs513

mjs513, 2018-04-05:
Oh no.  Sorry for the confusion.
 In the pop up window where you select what to plot on a chart and where you select the chart to plot I can't see all the variables (with the check boxes), the window goes off the screen.  I haven't been able to resize it or figure out how to see the rest of the variables.  Think I can see about 15 give or take of the variables.  Even the OK button is off the screen.

Lecerof, 2018-04-05:
I see, i figured it out when i tried it myself. I might be able to make a fix for it next week if you dont do it yourself. If you do, push it to a fork and I can pull it to my version of the program.

mjs513, 2018-04-05:
Unfortunately I don't know enough about Java to figure that one out.  Have enough challenges with c++.  If you do it let me know :)

mjs513, 2018-04-05:
I just looked at your fork - looks like you added a bunch of functionality that I wished Telemetry viewer had.  Nice Work!

mjs513, 2018-04-05:
Ok.  I give up - tried adding the scrollbar but no luck. probably adding it wrong:  This is what I added but it just gives me the header:
`		//------
		//JScrollPane scrollPane = new JScrollPane(windowContents);
		JScrollPane scrollPane=new JScrollPane(windowContents,
				ScrollPaneConstants.VERTICAL_SCROLLBAR_AS_NEEDED,
				ScrollPaneConstants.HORIZONTAL_SCROLLBAR_NEVER);
		setContentPane(scrollPane);
		scrollPane.setViewportView(windowContents);
		windowContents.setBorder(new EmptyBorder(10, 10, 10, 10));
		//windowContents.setLayout(new BoxLayout(windowContents, BoxLayout.Y_AXIS));
		windowContents.setPreferredSize(new Dimension(150, 410));
		add(windowContents);`