farrellf/TelemetryViewer #14: UI and layout save breaks while setting up charts

MiloMindbender, 2018-04-06:
This software is working great for me once it is setup.  However I am having problems with it while entering/editing the layout.  Sometimes after adding a new chart to the layout the save button stops working.  Ie you click it, it highlights, but the save dialogue does not appear.

Also, sometimes when creating/editing a chart the checkboxes next to some items in the dialog will not be there.

When the  program gets into this state sometimes it will let you save the layout but the file will be corrupt and I get an error when I try to reload it.

Workaround is to save really often and restart a lot (19 times on my last project) I can usually only add one or two charts before seeing this problem. 

Version issue?  Haven’t seen any other reports of this, maybe it’s me?

Windows 10 1706
Java 8. 161 (latest)
Running v4 jar from your website (not github)
4K monitor
2 nVidia quadro cards k5200 k4200 

It’s running perfect as long as I don’t bring up any dialogs.

If it helps I can try other java versions or test prerelease code.

MiloMindbender, 2018-08-22:
Near as I can tell, I've fixed this issue.  I found several different JREs running around in my system that I wasn't aware were there.  I went into add/remove programs and killed all the Javas I found there, then reinstalled Java 8 181 and JRE1.8.0_181 X86_64

After several hours of messing with this, V4 no longer seems to crash/freeze.  V5 also works fine.

I use this pretty heavily, if I don't see the issue again in a couple of days, I'll close it

farrellf, 2019-02-02:
It looks like multiple JREs was the problem since there have not been any comments in months. Closing this issue.

closed 2019-02-02 by farrellf