farrellf/TelemetryViewer #15: Weird open GL dialog on opening telemetry viewer

MiloMindbender, 2018-08-23:
This is on a Dell 5510 laptop which has both Intel graphics and Nvidia graphics built in.
Prior to testing, everything Java was uninstalled and Java 181 with JRE 181 was installed.
This is Windows 10 version 1709

When I launch the telemetry viewer I get a dialog box titled: OpenGL ES Error
and has the text "Can't find the name of Intel ICD OpenGL driver"
with an "ok" button.

If you click ok the box goes away briefly and comes back, if you click ok 8 times the Telemetry Viewer finishes launching and runs normally.

It doesn't stop you from using the viewer, it's just annoying.

farrellf, 2018-08-23:
Are you using a 32bit version of Java on a 64bit version of Windows? I've seen that error message when trying 32bit Java with 64bit Windows in a VM. I never found an exact answer as to why, but found some comments about JNI not working well when mixing 32bit and 64bit stuff.

MiloMindbender, 2018-09-11:
This is a 64 bit version of windows.   I uninstalled all my Java stuff first and reinstalled the base Java package and the 181 JRE 64 bit so there shouldn't be any 32 bit stuff out there.

Maybe this has something wto do with the dual graphics card in the laptop because installed it the same way on my desktop and no problems there.

farrellf, 2020-07-19:
I just released a new version, and one of the changes I made was upgrading to a pre-release version of the OpenGL library that I use. Please try the new version and see if this problem still occurs. More info here:

http://farrellf.com/projects/software/2020-07-17_Telemetry_Viewer_v0.7/

closed 2020-09-04 by farrellf