farrellf/TelemetryViewer #16: Possible to recieve broadcasted UDP messages?

fasse-sweden, 2018-08-30:
I'm connecting my PC to a ESP8266 i STA-mode, to be able to connect multiple PCs and recieve UDP messages I'm trying to use brodcast UDP. I'm not getting this to work, are there any reasons that Telemetry Viewer is not recieving these messages maybe they are filtered??

farrellf, 2018-08-30:
Broadcast UDP packets work for me, but I've only used my ESP8266 in client mode. Maybe use Wireshark to see if the packets are being received by your PC?

MiloMindbender, 2018-08-31:
Check your pc’s firewall settings, mine initially was blocking java so other programs received udp and. Java did not. There might be blocks on broadcast traffic too

BcGh01, 2018-09-03:
Hi MiloMindbender. Sorry if this is not the right place to ask but you seem very knowledgeable and I could really do with some help. I have a Windows 10 PC and have loaded Java runtime and the development environment. I have also install the Arduino software. I downloaded the TelemetryViewer V0.5 and it shows as an executable java file in Explorer. However, when I double click on ir nothing happens ! Can you help please.

MiloMindbender, 2018-10-26:
Hi BcGh01, sorry I didn't see your question sooner.  If your telemetryviewer jar file shows as a java file (check properties->general tab , should say "opens with: Java(TM) Platform SE") then it should just work.

If it's not, it is possible there is some problem in your Java install.  When I was having problems I checked  "add/remove programs" and found 3 other versions of java besides 8 were installed.  Take a look and see what you have installed now, if you see a lot of versions, I'd uninstall all the java stuff and then just reinstall the latest Java 8 and JRE.  Also make sure you install the right word size (32 or 64 bit) for your OS.  This fixed all my problems on two machines.

Hope it helps you.