farrellf/TelemetryViewer #17: Will Telemetry Viewer work on Raspberry PI?

MiloMindbender, 2018-12-09:
I wanted to install TelemetryViewer on a Raspberry PI so I could just plug it into whatever TV was handy and look at telemetry there.  I have the standard Java that's included with PI linux and get these errors when I try to start up.

It appears to be complaining about the GL version but I'm afraid I'm not enough of a Java guy to understand exactly what's wrong.  Can anyone help?

libEGL warning: DRI2: failed to authenticate
Exception in thread "main" java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:58)
Caused by: java.lang.InternalError: Available GLVersions not set for EGLGraphicsDevice[type .egl, v1.4.0, connection decon, unitID 0, handle 0x60feb8f8, owner true, NullToolkitLock[obj 0xc88a32]]
	at com.jogamp.opengl.GLProfile.initProfilesForDeviceCritical(GLProfile.java:1952)
	at com.jogamp.opengl.GLProfile.initProfilesForDevice(GLProfile.java:1875)
	at com.jogamp.opengl.GLProfile.initProfilesForDefaultDevices(GLProfile.java:1842)
	at com.jogamp.opengl.GLProfile.access$000(GLProfile.java:80)
	at com.jogamp.opengl.GLProfile$1.run(GLProfile.java:230)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.jogamp.opengl.GLProfile.initSingleton(GLProfile.java:216)
	at com.jogamp.opengl.GLProfile.getProfileMap(GLProfile.java:2297)
	at com.jogamp.opengl.GLProfile.get(GLProfile.java:988)
	at com.jogamp.opengl.GLProfile.get(GLProfile.java:1004)
	at OpenGLChartsRegion.<init>(OpenGLChartsRegion.java:99)
	at Main.main(Main.java:18)
	... 5 more

farrellf, 2018-12-10:
If I recall correctly, the Pi only supports OpenGL ES. TelemetryViewer requires "normal" OpenGL (not ES) because it does not use the shader-based pipeline. I have plans to eventually move over to shaders and support OpenGL ES, but that is a low priority and I have very little free time these days.

cooked, 2019-06-05:
Hi @farrellf ,
I'm also looking at running TelemetryViewer on the Raspberry. Could you anticipate if at some point you will work on the porting or could you make a list of the parts of code that need to be rewritten?
cheers

farrellf, 2019-06-05:
I'd like to get around to it within a year, but other things are a higher priority for me, so it's hard to give an accurate estimate.

If you'd like to work on it, you'd basically need to rewrite all of the OpenGL code, which is mostly in the OpenGL*.java files. I also don't know if the font rendering code requires regular OpenGL or is compatible with OpenGL ES.

It'll be a significant task to undertake.

farrellf, 2020-07-19:
I just released a new version, and it works on the Pi 4. More info here:
http://farrellf.com/projects/software/2020-07-17_Telemetry_Viewer_v0.7/

For now, it only works on the Pi 4. I am looking into how difficult it would be to support earlier Pi's, and it seems like that would require more work than I can justify right now.

closed 2020-09-05 by farrellf