farrellf/TelemetryViewer #26: Support for 32 bit int, 64 bit float (binary) would be nice

MiloMindbender, 2019-05-24:
I have some high precision IMUs (accelerometer, gyro) that output 32 bit int or 64 bit (double) float format, it would be nice to have support for either of these formats.  These IMUs actually need that much precision, without it you can't see the low level noise when the device is sitting still.

One of the IMUs returns a 32 bit integer, which you have to convert to double and multiply by 0.000244148075808 to get acceleration in Gs. Has to be done with doubles or the low-level noise disappears.  Would be nice to be able to take a 32 bit int and apply this conversion in TelemetryViewer

farrellf, 2019-05-24:
I plan to implement this in the future, but it will require significant work to support. Don't expect it any time soon. Right now everything is converted into 32bit floats, because that's quick and easy for GPUs to process, and because it keeps the code simple.

farrellf, 2021-07-25:
The latest release added support for uint32, but everything still gets converted into float32's so there can be a loss of precision. I still plan to improve this in the future.

strud, 2023-02-04:
Would it be much more work to add support for int32?   I would dearly like to capture 32bit encoder counter values that are signed.

farrellf, 2023-02-07:
Support for int32 has been added, it will be in the next release. But the code still converts everything to float32's, so the possible loss of precision still exists.