farrellf/TelemetryViewer #27: log string data and print as lable

binarymaker, 2019-06-14:
just read csv string or number value and print as lable

i requited print hex values and message strings.

Ex: printf("%d,%s\n\r", timestamp, "lamp off")

farrellf, 2019-06-14:
So you just want the telemetry printed out to the console? This software is for visualizing the data, not just printing out text. You might be better served with a terminal program or small custom-written program.

binarymaker, 2019-06-15:
I want both options. Some data no need to plot.
ex. status ON/OFF, warning messages.

if you develop, it will more use full for all type of projects

Thanks

farrellf, 2019-06-15:
The next version of TelemetryViewer will have better support for timestamps and for boolean/enum telemetry. But I have no plans to implement what you're asking for. Samples can be exported to a CSV file for later analysis, but the main goal of this program is for visualizing a stream of telemetry data.

closed 2019-06-17 by binarymaker