farrellf/TelemetryViewer #28: Telemetry Viewer Stops logging.

OEQ6, 2019-08-01:
Telemetry viewer I believe runs out of buffer space and stops logging and freezes. Is it possible to log that data and save it after buffer reaches near limit or a set limit then clears buffer, save and continue logging.This only gives me a few hours of logging. I'm trying to fix the issue but I've never coded in Java before. 

Thank you.

farrellf, 2019-08-02:
Right now all samples are stored in RAM. I plan to support offloading to disk, so truly massive datasets can be acquired, but it might be a while before I get to that.

On a PC with 8GB of RAM I regularly acquire ~30 million samples, each sample composed of 10 datasets. After how many samples do you start having problems? How much RAM do you have?

OEQ6, 2019-08-02:
I also have 8 gigs. This one stops at 938978 almost a million. Sometime it
stops at less.

farrellf, 2019-08-02:
Weird. Perhaps an Exception happens. Can you try running from the command line? (java -jar file.jar) If there is an Exception, a stacktrace should get printed.

OEQ6, 2019-08-02:
Ok. I tried and nothing happened. It just opened the viewer.

OEQ6, 2019-08-02:
I'm trying to add the auto save feature back. But I not well versed in
java. Plan is to auto save the buffer then clear it, and have the telemetry
viewer view the last hour of data recordings in the logged file.

Would it be possible for the program to clear the buffer before it starts.
Which will make more room for the data received.

farrellf, 2019-09-18:
This issue should be solved now. Version 0.6 swaps data to disk when RAM runs low, worked around a memory leak in TextRenderer, and fixed some issues in the Time Domain Chart when the sample number was very large. And if you export a CSV log, it now contains UNIX timestamps for each sample.

Relevant commits are:
https://github.com/farrellf/TelemetryViewer/commit/0e95457c909a8c52c079c27b1515a1465eab299c
https://github.com/farrellf/TelemetryViewer/commit/bbe18fdae7b1a8cb005d1ee9e9625c209939ccce
https://github.com/farrellf/TelemetryViewer/commit/841c21ad2e032cfa5fde93b19344876cf8e69175
https://github.com/farrellf/TelemetryViewer/commit/f27c24c22822f7246dcc49093b810f1419cda989
https://github.com/farrellf/TelemetryViewer/commit/26ae7538c275213b58284e9dba1e063485f21d82

Please try out v0.6, and if things seems fine, close this issue.
http://farrellf.com/TelemetryViewer/TelemetryViewer_v0.6.jar

closed 2020-06-19 by farrellf