farrellf/TelemetryViewer #30: Exception in thread "main" java.lang.reflect.InvocationTargetException

Gord1, 2019-11-22:
Hi,
I'm eager to try your Telemetry Viewer, but have run into a problem.  This is what I get and sorry I don't know anything about Java.  If I can provide more info, let me know.
Gord_W

gordon@linuxAMD:~$ java -jar /home/gordon/Downloads/TelemetryViewer_v0.6.jar
Exception in thread "main" java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:61)
Caused by: java.lang.InternalError: XXX0 profile[1]: GL3bc -> profileImpl GL4bc !!! not mapped 
	at com.jogamp.opengl.GLProfile.computeProfileMap(GLProfile.java:2071)
	at com.jogamp.opengl.GLProfile.initProfilesForDeviceCritical(GLProfile.java:1954)
	at com.jogamp.opengl.GLProfile.initProfilesForDevice(GLProfile.java:1875)
	at com.jogamp.opengl.GLProfile.initProfilesForDefaultDevices(GLProfile.java:1842)
	at com.jogamp.opengl.GLProfile.access$000(GLProfile.java:80)
	at com.jogamp.opengl.GLProfile$1.run(GLProfile.java:230)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.jogamp.opengl.GLProfile.initSingleton(GLProfile.java:216)
	at com.jogamp.opengl.GLProfile.getProfileMap(GLProfile.java:2297)
	at com.jogamp.opengl.GLProfile.get(GLProfile.java:988)
	at com.jogamp.opengl.GLProfile.get(GLProfile.java:1004)
	at OpenGLChartsRegion.<init>(OpenGLChartsRegion.java:108)
	at Main.main(Main.java:31)
	... 5 more

I'm running Linux Mint with the latest update of Java V8.

System:
System:    Host: linuxAMD Kernel: 4.15.0-66-generic x86_64 bits: 64 compiler: gcc v: 7.4.0 
           Desktop: Cinnamon 4.2.4 wm: muffin dm: LightDM Distro: Linux Mint 19.2 Tina 
           base: Ubuntu 18.04 bionic 
Machine:   Type: Desktop Mobo: MSI model: FM2-A55M-E33 (MS-7721) v: 2.0 serial: <filter> 
           UEFI: American Megatrends v: 11.6 date: 02/12/2015 
CPU:       Topology: Quad Core model: AMD A8-5600K APU with Radeon HD Graphics bits: 64 type: MCP 
           arch: Piledriver rev: 1 L2 cache: 2048 KiB 
           flags: lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm bogomips: 28746 
           Speed: 1519 MHz min/max: 1400/3600 MHz Core speeds (MHz): 1: 1397 2: 1397 3: 1396 
           4: 1397 
Graphics:  Device-1: AMD Trinity [Radeon HD 7560D] vendor: Micro-Star MSI driver: radeon v: kernel 
           bus ID: 00:01.0 chip ID: 1002:9904 
           Display: x11 server: X.Org 1.19.6 driver: ati,radeon unloaded: fbdev,modesetting,vesa 
           resolution: 1920x1080~60Hz 
           OpenGL: renderer: AMD ARUBA (DRM 2.50.0 / 4.15.0-66-generic LLVM 9.0.0) 
           v: 4.3 Mesa 19.3.0-devel (git-ffb0d3a 2019-09-29 bionic-oibaf-ppa) compat-v: 3.1 
           direct render: Yes

farrellf, 2019-11-22:
That exception occurred when trying to get an OpenGL 2.0 profile. It looks like your Radeon HD 7560D supports OpenGL 4.4, so it should work. I suspect there may be an issue with your graphics drivers. Are you able to run any other programs that use OpenGL?

Check if there is a new version of the drivers available, or if you're using some open source drivers, try the official drivers instead.

I tried running in an Ubuntu VM, with Intel integrated graphics, and it worked for me.

Gord1, 2019-11-22:
Hi,
glxgears opengl test works fine.  I use a cad program that requires opengl.

Here is some info on my system.  There is a line with OpenGL profile below.  Is that not good enough?
Thanks
Gord_W

gordon@linuxAMD:~$ glxinfo
name of display: :0
display: :0  screen: 0
direct rendering: Yes
server glx vendor string: SGI
server glx version string: 1.4
server glx extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_profile, 
    GLX_ARB_create_context_robustness, GLX_ARB_fbconfig_float, 
    GLX_ARB_framebuffer_sRGB, GLX_ARB_multisample, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_libglvnd, GLX_EXT_texture_from_pixmap, 
    GLX_EXT_visual_info, GLX_EXT_visual_rating, GLX_INTEL_swap_event, 
    GLX_MESA_copy_sub_buffer, GLX_OML_swap_method, GLX_SGIS_multisample, 
    GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, GLX_SGIX_visual_select_group, 
    GLX_SGI_make_current_read, GLX_SGI_swap_control
client glx vendor string: Mesa Project and SGI
client glx version string: 1.4
client glx extensions:
    GLX_ARB_context_flush_control, GLX_ARB_create_context, 
    GLX_ARB_create_context_no_error, GLX_ARB_create_context_profile, 
    GLX_ARB_create_context_robustness, GLX_ARB_fbconfig_float, 
    GLX_ARB_framebuffer_sRGB, GLX_ARB_get_proc_address, GLX_ARB_multisample, 
    GLX_EXT_buffer_age, GLX_EXT_create_context_es2_profile, 
    GLX_EXT_create_context_es_profile, GLX_EXT_fbconfig_packed_float, 
    GLX_EXT_framebuffer_sRGB, GLX_EXT_import_context, 
    GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, GLX_EXT_visual_rating, 
    GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, 
    GLX_MESA_multithread_makecurrent, GLX_MESA_query_renderer, 
    GLX_MESA_swap_control, GLX_OML_swap_method, GLX_OML_sync_control, 
    GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
    GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control, GLX_SGI_video_sync
GLX version: 1.4
GLX extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_profile, 
    GLX_ARB_create_context_robustness, GLX_ARB_fbconfig_float, 
    GLX_ARB_framebuffer_sRGB, GLX_ARB_get_proc_address, GLX_ARB_multisample, 
    GLX_EXT_buffer_age, GLX_EXT_create_context_es2_profile, 
    GLX_EXT_create_context_es_profile, GLX_EXT_fbconfig_packed_float, 
    GLX_EXT_framebuffer_sRGB, GLX_EXT_import_context, 
    GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, GLX_EXT_visual_rating, 
    GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, GLX_MESA_query_renderer, 
    GLX_MESA_swap_control, GLX_OML_swap_method, GLX_OML_sync_control, 
    GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
    GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control, GLX_SGI_video_sync
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: X.Org (0x1002)
    Device: AMD ARUBA (DRM 2.50.0 / 4.15.0-66-generic, LLVM 9.0.0) (0x9904)
    Version: 19.3.0
    Accelerated: yes
    Video memory: 768MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 4.3
    Max compat profile version: 3.1
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.1
Memory info (GL_ATI_meminfo):
    VBO free memory - total: 767 MB, largest block: 767 MB
    VBO free aux. memory - total: 1021 MB, largest block: 1021 MB
[... 253 lines cut ...]
    GL_KHR_texture_compression_astc_sliced_3d, 
    GL_MESA_shader_integer_functions, GL_NV_conditional_render, 
    GL_NV_draw_buffers, GL_NV_fbo_color_attachments, GL_NV_image_formats, 
    GL_NV_read_buffer, GL_NV_read_depth, GL_NV_read_depth_stencil, 
    GL_NV_read_stencil, GL_OES_EGL_image, GL_OES_EGL_image_external, 
    GL_OES_EGL_image_external_essl3, GL_OES_EGL_sync, 
    GL_OES_compressed_ETC1_RGB8_texture, GL_OES_copy_image, GL_OES_depth24, 
    GL_OES_depth_texture, GL_OES_depth_texture_cube_map, 
    GL_OES_draw_buffers_indexed, GL_OES_draw_elements_base_vertex, 
    GL_OES_element_index_uint, GL_OES_fbo_render_mipmap, 
    GL_OES_geometry_point_size, GL_OES_geometry_shader, 
    GL_OES_get_program_binary, GL_OES_gpu_shader5, GL_OES_mapbuffer, 
    GL_OES_packed_depth_stencil, GL_OES_primitive_bounding_box, 
    GL_OES_required_internalformat, GL_OES_rgb8_rgba8, GL_OES_sample_shading, 
    GL_OES_sample_variables, GL_OES_shader_image_atomic, 
    GL_OES_shader_io_blocks, GL_OES_shader_multisample_interpolation, 
    GL_OES_standard_derivatives, GL_OES_stencil8, GL_OES_surfaceless_context, 
    GL_OES_tessellation_point_size, GL_OES_tessellation_shader, 
    GL_OES_texture_3D, GL_OES_texture_border_clamp, GL_OES_texture_buffer, 
    GL_OES_texture_cube_map_array, GL_OES_texture_float, 
    GL_OES_texture_float_linear, GL_OES_texture_half_float, 
    GL_OES_texture_half_float_linear, GL_OES_texture_npot, 
    GL_OES_texture_stencil8, GL_OES_texture_storage_multisample_2d_array, 
    GL_OES_texture_view, GL_OES_vertex_array_object, GL_OES_vertex_half_float, 
    GL_OES_viewport_array

Gord1, 2019-11-25:
I removed the development version of Mesa and installed the current Linux Mint distribution version and Telemetry Viewer now works.  Go figure ...  OpenGL core profile is 4.3.  
Issue is closed.
Thanks for your time,
Gord_W

gordon@linuxAMD:~$ glxinfo | grep OpenGL
OpenGL vendor string: X.Org
OpenGL renderer string: AMD ARUBA (DRM 2.50.0 / 4.15.0-66-generic, LLVM 8.0.0)
OpenGL core profile version string: 4.3 (Core Profile) Mesa 19.0.8
OpenGL core profile shading language version string: 4.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.1 Mesa 19.0.8
OpenGL shading language version string: 1.40
OpenGL context flags: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.1 Mesa 19.0.8
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.10
OpenGL ES profile extensions:

closed 2019-11-25 by Gord1

fattahalf, 2020-07-07:
Hello Gord1. I got the same issue. How to solve it? I am currently working on Ubuntu 20.04