farrellf/TelemetryViewer #33: Again - Exception in thread "main" java.lang.reflect.InvocationTargetException

Gord1, 2020-03-19:
Reference: [https://github.com/farrellf/TelemetryViewer/issues/30](url)
Hi,
Back in November of last year I ran into this problem (link above).  To recap, I was getting a crash when running TelemetryViewer on startup.  I was using a development version of Mesa (v: 4.3 Mesa 19.3.0-devel (git-ffb0d3a 2019-09-29 bionic-oibaf-ppa) compat-v: 3.1) but found that if I used the standard Linux Mint version at that time of Mesa 19.0.8 the problem went away.

Time has marched on and the standard Linux Mint version is now Mesa 19.2.8 using the automatic updates.  I now have the same problem.

I would guess that anyone else that tries TelemetryViewer on Linux would have the same problem.

Something between Mesa 19.0.8 and 19.2.8 has changed and causes the problem.

In the mean time I'm going to try and roll back Mesa to 19.0.8 as a work around.
Regards,
Gord_W

Gord1, 2020-03-19:
This is the up-to-date Linux Mint that is causing problems:

gordon@linuxAMD:~$ glxinfo | grep OpenGL
OpenGL vendor string: X.Org
OpenGL renderer string: AMD ARUBA (DRM 2.50.0 / 4.15.0-91-generic, LLVM 9.0.0)
OpenGL core profile version string: 4.3 (Core Profile) Mesa 19.2.8
OpenGL core profile shading language version string: 4.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.1 Mesa 19.2.8
OpenGL shading language version string: 1.40
OpenGL context flags: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.1 Mesa 19.2.8
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.10
OpenGL ES profile extensions:

Gord1, 2020-03-19:
I thought that I could revert back to Mesa 19.0.8 using TimeShift, but I don't have a backup in late Nov.  I'm stuck.  I'm a noob.
Gord_W

farrellf, 2020-07-19:
I just released a new version, and I think it will fix this problem. I updated the OpenGL library that I use to a newer pre-release version, and I also rewrote most of my OpenGL code. I even wrote my own font rendering code because the font rendering library I was using had some bugs.

http://farrellf.com/TelemetryViewer/TelemetryViewer_v0.7.jar
http://farrellf.com/projects/software/2020-07-17_Telemetry_Viewer_v0.7/

Telemetry Viewer now requires OpenGL 3.2 Core Profile, or OpenGL ES 3.1 with the GL_EXT_geometry_shader extension.

Gord1, 2020-07-27:
Awesome!!  That seems to work well so far.  This closes the issue. Thanks!

closed 2020-07-27 by Gord1