farrellf/TelemetryViewer #34: suggestion: digital time clock lookalike card

thostie2002, 2020-04-03:
i use a real time clock module to import time to the telemetry viewer, now i use 3 dail chards one for hours, one for minutes, one for seconds see the first image.

please can you make a simple digital clock chard? like the second image?

thanks:)

first image
[image]

second image
[image]

farrellf, 2020-09-05:
I agree, it would be nice to have a simple numeric display. It's on my to do list ;)

farrellf, 2021-07-25:
The latest release now supports this. There is a "Statistics Chart" and it can be configured to show just the current values:

http://www.farrellf.com/projects/software/2021-07-24_Telemetry_Viewer_v0.8/
https://youtu.be/NUBm9b8YQMQ?t=646

closed 2021-08-18 by farrellf