farrellf/TelemetryViewer #36: Error when execute Telemetry Viewer on Ubuntu 20.04

fattahalf, 2020-07-07:
Hello. Great work on Telemetry Viewer. Such a good software.
I recently work on Ubuntu 20.04, and when i try to execute .jar file of Telemetry Viewer 0.6, i got this message :

Exception in thread "main" java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:61)
Caused by: com.jogamp.opengl.GLException: Profile GL2 is not available on null, but: [GLProfile[GLES1/GLES1.hw], GLProfile[GLES2/GLES3.hw], GLProfile[GL2ES1/GLES1.hw], GLProfile[GL4ES3/GL4.hw], GLProfile[GL2ES2/GL4.hw], GLProfile[GL4/GL4.hw], GLProfile[GLES3/GLES3.hw], GLProfile[GL4/GL4.hw], GLProfile[GL3/GL4.hw], GLProfile[GL2GL3/GL4.hw]]
	at com.jogamp.opengl.GLProfile.get(GLProfile.java:991)
	at com.jogamp.opengl.GLProfile.get(GLProfile.java:1004)
	at OpenGLChartsRegion.<init>(OpenGLChartsRegion.java:108)
	at Main.main(Main.java:31)
	... 5 more

I have no idea about this error. Do you have any ideas to solve this problem? It would be great if you have one. Thank you.

farrellf, 2020-07-07:
That exception basically means it could not get an OpenGL 2 context. What graphics card are you using?

fattahalf, 2020-07-07:
[image]

fattahalf, 2020-07-07:
I am using Intel HD Graphics 520

Gord1, 2020-07-07:
Is this the same problem in
Ref. https://github.com/farrellf/TelemetryViewer/issues/30
Which was solved by going back to an older Mesa driver.
Unfortunately the older Mesa driver is no longer available and none of the newer drivers work.
Ref. https://github.com/farrellf/TelemetryViewer/issues/33

I'm using Linux Mint 19.3 and that is based off of Ubuntu.  So if you find a fix for @fattahalf , that will likely solve my problem too.

Gord_W

fattahalf, 2020-07-19:
Thank you Gord_W for the answer

closed 2020-07-19 by fattahalf

Gord1, 2020-07-19:
Were you able to getting it working somehow or did you just give up because it is broken on Linux?

Gord_W

farrellf, 2020-07-19:
I just released a new version, and I think it will fix this problem. I updated the OpenGL library that I use to a newer pre-release version, and I also rewrote most of my OpenGL code. I even wrote my own font rendering code because the font rendering library I was using had some bugs.

http://farrellf.com/TelemetryViewer/TelemetryViewer_v0.7.jar
http://farrellf.com/projects/software/2020-07-17_Telemetry_Viewer_v0.7/

Telemetry Viewer now requires OpenGL 3.2 Core Profile, or OpenGL ES 3.1 with the GL_EXT_geometry_shader extension.