farrellf/TelemetryViewer #37: Display delay when using UDP.

YutaSuzu, 2020-08-10:
Hi, I’m trying to using UDP with CSV.
When I receive data at first connection. Always fine.
But once when I tried save or disconnect and reconnect.
Display of telemetry having lag.
When press disconnect, sometime appear “UDP packet error”
Did you notice this phenomena. 

Serial UART connection with CSV is fine. No problem.

My system condition is 
Windows10 Version 1909 (OS Build 18363.959)
Java Version 8 (build 1.8.0_251-b08)

UDP packet transfer with Arduino mkr1000 WifiUDP.h.

Yutaka Suzuki

farrellf, 2020-09-05:
Hmm, I was able to duplicate at least part of this problem when under heavy load. When I tested with <1000 packets per second it worked fine, but when I flooded it with ~200,000 packets per second I found several problems.

I'm working on this, and hope to have it fixed in the next version.

SolidGeek, 2021-04-17:
I am having similar issues. I have two computers, my main PC and a laptop both listening on a WiFi network. The UPD packages are received at the same frequency on both devices (checked with Wireshark), however, the laptop has a large amount of delay / the data is read much slower in the Telemetry Viewer. A couple of times the laptop has worked similar to the PC, however not often. The plots on the laptop only move a few samples per second, while on the PC they move at 200 samples per second (the transmit frequency). The weird thing is that if I disconnect the WiFi, the Telemetry Viewer charts keep updating with measurements on my laptop like they are stuck in some sort of buffer. Also when I press "Disconnect" I get the notification "UDP packet error".

Both devices are running the latest Windows, Java and OpenJDK. Any idea what might be the problem?

SolidGeek, 2021-04-17:
@farrellf I have tried the Telemetry Viewer on two different laptops, however, none of them performs well. Only very few samples are loaded into the plots each second. I have tried running the software through Eclipse, to see if any errors were happening, but nothing comes up. The UDP packages are received at 200Hz in Wireshark, thus I do not understand why they are not loaded by the Telemetry Viewer. Both laptops have powerful CPUs and GPUs, so I don't think it is a performance issue. Any idea what could cause this?

farrellf, 2021-07-25:
The latest release should fix these problems. The data processing code was completely rewritten. Some bugs were fixed and it is also much more efficient. My laptop can easily process and visualize >5Gbps of telemetry data now.

http://www.farrellf.com/projects/software/2021-07-24_Telemetry_Viewer_v0.8/
https://www.youtube.com/watch?v=NUBm9b8YQMQ&t=1317s

YutaSuzu, 2021-08-21:
Hello. I just use new version of Telemetry Viewer. 
Yes!! it has been fixed problem. I also loved new function!!
I will close this issue.
Thank you very much. 

Yutaka Suzuki

closed 2021-08-21 by YutaSuzu