farrellf/TelemetryViewer #38: [Feature Request] Remember Last Used Parameters (With Port, Server, Telemetry Params)

ajaybnl, 2020-09-09:
I use this Great Software daily for my Esp8266 and Esp32 UDP Charts.

But, I need to manually Enter Port, UDP etc from the First Dialog & Afterwards fill the two required fields daily 10+ Times. 

I just want to launch it with all things pre loaded from last used settings.

Please make this a Feature.

Brad-HES, 2022-06-24:
This would simply be setting up the main to automatically import a configuration file at startup. The code already is able to import configuration files, so it should just be a matter of making it do this at startup.

Masresha-y, 2022-06-25:
can you help adding google map and identifying a location and waypoints
please thank you

closed 2022-11-20 by ajaybnl