farrellf/TelemetryViewer #40: Version 0.7 hangs at startup on macOS High Sierra (10.13.6)

jimsmart, 2020-09-22:
Hi,

Great project, thanks!

I am trying to run the v0.7 jar, but it hangs (with a spinning beachball) after creating the main window. It's hanging before it manages to resize the window to fullscreen and before painting any actual content, if that is of any help?

There are no errors reported in the console, only the message:

```
[HINT   ]   Start by connecting to a device or opening a file by using the buttons below.
```

I've tried various JREs/JDKs, and get the same results.

I am able to run v0.6 (and earlier) without any issues.

— Have you any suggestions please?

Cheers.

farrellf, 2020-09-22:
In v0.7 I updated the OpenGL library, and it sounds like there is a compatibly problem with OS X. Unfortunately I don't have access to any Macs right now for testing. I will try to find out if there are any work arounds or solutions. It would help to know if this problem only occurs with High Sierra or also with newer/older versions. Do you have access to other Macs that you could test on?

greenonline, 2020-12-10:
Just to add, I get the _exact_ same issue (again on 10.13.6). 

FWIW, the JDK I'm using was installed via HomeBrew:  

`1.8.0_275, x86_64:	"AdoptOpenJDK 8"	/Library/Java/JavaVirtualMachines/adoptopenjdk-8.jdk/Contents/Home`

TBH, I'm glad to find that I'm not the only one with the same issue. Been scratching my head about it for a few days now.

I might be able to test it on some older versions of OS X, but nothing later than High Sierra (10.13.6). Let me know if that would be useful.

farrellf, 2020-12-11:
Since the Jogamp/GlueGen files from v0.6 seem to fix the problem in v0.7, I'd almost guarantee that this is due to a bug in the prerelease version I used in v0.7.

I had to use a prerelease version because the stable version has not had a new release in years, and it has several compatibility problems of its own (especially if you want to use Java 9 or newer.)

At this point I have to wait for the Jogamp team to make progress and provide another prerelease or release build.

farrellf, 2021-01-17:
There's a new prerelease build of JOGL. Can you try it and see if it fixes the problem?

https://jogamp.org/deployment/v2.4.0-rc-20210111/jar/
https://jogamp.org/deployment/v2.4.0-rc-20210111/fat/

Gadgeteering, 2022-03-09:
Maybe another workaround is java --add-exports=java.base/java.lang=ALL-UNNAMED --add-exports=java.desktop/sun.awt=ALL-UNNAMED --add-exports=java.desktop/sun.java2d=ALL-UNNAMED -jar TelemetryViewer_v0.7.jar