farrellf/TelemetryViewer #43: Problem in WidgetHistogramYaxisType.java - Unlikely argument type for equals(): String seems to be unrelated to JComboBox<String>

greenonline, 2020-12-10:
Eclipse is showing me that there is a "problem" (rather than an error) here: https://github.com/farrellf/TelemetryViewer/blob/master/Telemetry%20Viewer/src/WidgetHistogramYaxisType.java#L243

```
Description Resource Path Location Type
Unlikely argument type for equals(): String seems to be unrelated to JComboBox<String> WidgetHistogramYaxisType.java /TelemetryViewer-mgj2/src line 243 Java Problem
```

The line is

```
if(axisTypeCombobox.equals(type))
```

I *think* that it should be

```
if(axisTypeCombobox.getSelectedItem().toString().equals(type))
```

but I may be wrong. .? Either way that types don't match in the `equals()`. 

There's a reason why it compiles without error, but the problem still shows up, which _seems_ to be explained in [this answer](https://stackoverflow.com/a/54756028/4424636) to the question [Elegant way to overcome Unlikely argument type for equals(): Stream<String> seems to be unrelated to String](https://stackoverflow.com/q/54746294/4424636).

farrellf, 2020-12-11:
You're right, it should be:

`if(axisTypeCombobox.getSelectedItem().toString().equals(type))`

I'll fix this and include it in the next release. Thanks.

farrellf, 2021-07-25:
The fix is in the latest release:
http://www.farrellf.com/projects/software/2021-07-24_Telemetry_Viewer_v0.8/

closed 2021-08-18 by farrellf