farrellf/TelemetryViewer #45: [Feature Request] Time axis from timestamp in the data?

MiloMindbender, 2020-12-10:
In some of my applications the data doesn't come out at a fixed frequency.  However the data source has a real time clock and can put a time stamp in the data it sends.

Could the graphs be setup to use this time stamp as the X axis of charts?  This would be very useful when the data doesn't come on a regular schedule.  The device can send time in just about any format with microsecond accuracy.  It could be sent as real time in some format or something simpler like seconds (or microseconds) since startup in either a float or integer format.

I've been using your tool for some time and it is really great!  My latest project is using it to visualize 3d camera tracking data from an HTC VIVE VR tracking system.  In this setup the data can come at a fixed sampling rate but in some cases the sampling rate is variable based on the 3D scene being visualized.

farrellf, 2020-12-11:
I plan to add a scatter plot visualization, with the ability to draw lines between coordinates. That should give you the desired effect (just feed it something like a UNIX timestamp for the x-axis...) I hope to implement this in time for the next release, but I don't have much spare time these days. It might be half a year or more before I can do this.

MiloMindbender, 2020-12-14:
Don't really need a scatter plot, just the existing chart but with the x axis being time would be fine.