farrellf/TelemetryViewer #46: Export doesn't work till some data is captures

MiloMindbender, 2020-12-14:
Maybe I'm missing something, but it seems like the "export" button isn't available till some data is captured.  It would be nice if it was available all the time so that when you enter the layout for a new telemetry message you could save it right away without having to capture some data first.

I think in older versions export was always enabled.

farrellf, 2020-12-15:
I agree, I'll modify the export functionality in time for the next release.

Driver55, 2021-04-27:
I use TelemetryViewer v.07 in windows 10 and the crash also happens simply when the transmission speed is different between the controlled unit and TelemetryViewer.
Again there is no way to disconnect to change the baud rate. The only possibility is to abort telemetry tasks

farrellf, 2021-07-25:
The latest release has this change. As for the problem with disconnecting, that should also be fixed because almost all of the connection and data processing code was rewritten (fixing some bugs in the process.)

closed 2021-08-18 by farrellf