farrellf/TelemetryViewer #49: Change STL model for Quaternion Chart

SolidGeek, 2021-04-15:
I am using the program to visualize the results of my university project (a single motor UAV), transmitted by an ESP32 over UDP. Works flawlessly! For the later presentation of the project (the exam), it would be nice to be able to change the STL of the monkey to something else (3D-CAD model of my drone). 

I have tried downloading your project and built a version with a different STL file using Eclipse. However, after building the project, nothing is showing in the quaternion chart (even though data is still being received in the other charts). I am new to developing Java, thus I have not a single clue what might be the problem. Does the STL file need to be in some specific format for it to work?  Maybe I am missing some build settings?

Hope you would consider adding this feature or helping me out!
Thanks for your great work

andresrobinson, 2021-04-15:
I still thing it would be a great addition if this replacement could be done in the GUI

SolidGeek, 2021-04-15:
> I still thing it would be a great addition if this replacement could be done in the GUI

I agree, however, the way the STL rendering is happening right now does not handle larger files well. The 3D rendering has to be re-done or optimized in order to have any real value, as any STL above 1 Mb crashes the software. Also if the STL is not scaled right nothing shows. Lots of potential problems 👍

farrellf, 2021-04-16:
I agree, this would be a good feature. It's on my ever growing to-do list :)

merjul, 2024-08-09:
I have already tried rescaling within a bounding box from -1 to 1 as @SolidGeek  said, but still nothing shows up in the quanternion plot. What I have done so far:

- Rescaled my STL model within a bounding box from -1 to 1
- Export the STL model as ASCII type
- Renamed my STL model to monkey.stl
- Replace the existing monkey.stl within the telemetry viewer file.

Are there any steps missing? because I am not familiar with the handling of these STL's. Any help would be much appreciated.

Salam797, 2025-08-10:
@SolidGeek / @farrellf . Could you please walk me through how to load a custom stl? Your help and assistance is greatly appreciated.