farrellf/TelemetryViewer #55: No tiles in version 0.8

Waschbaerbauch, 2021-09-26:
Hello!

I try to use version 0.8 but when I setup the connection I have no tiles.
I'm using java version "1.8.0_301" Win 10 home 64bit.
With versin 0.6 it is working.

best regards

Stefan

farrellf, 2021-09-28:
Hi. Version 0.8 used some features of Java 9. If you upgrade to Java 9 or newer it should work properly.

danmo24, 2021-10-01:
Same problem.

I downloaded the newest JRE
Java(TM) SE Runtime Environment (build 1.8.0_301-b09)

There is no newer Version of java available?!

Log shows:
`Exception in thread "main-AWTAnimator#00" com.jogamp.opengl.util.AnimatorBase$UncaughtAnimatorException: java.lang.RuntimeException: com.jogamp.opengl.GLException: Caught NoSuchMethodError: java.nio.FloatBuffer.rewind()Ljava/nio/FloatBuffer; on thread AWT-EventQueue-0
        at com.jogamp.opengl.util.AWTAnimatorImpl.display(AWTAnimatorImpl.java:92)
        at com.jogamp.opengl.util.AnimatorBase.display(AnimatorBase.java:453)
        at com.jogamp.opengl.util.Animator$MainLoop.run(Animator.java:204)
        at java.lang.Thread.run(Unknown Source)
Caused by: java.lang.RuntimeException: com.jogamp.opengl.GLException: Caught NoSuchMethodError: java.nio.FloatBuffer.rewind()Ljava/nio/FloatBuffer; on thread AWT-EventQueue-0
        at com.jogamp.common.util.awt.AWTEDTExecutor.invoke(AWTEDTExecutor.java:58)
        at jogamp.opengl.awt.AWTThreadingPlugin.invokeOnOpenGLThread(AWTThreadingPlugin.java:103)
        at jogamp.opengl.ThreadingImpl.invokeOnOpenGLThread(ThreadingImpl.java:201)
        at com.jogamp.opengl.Threading.invokeOnOpenGLThread(Threading.java:202)
        at com.jogamp.opengl.Threading.invoke(Threading.java:221)
        at com.jogamp.opengl.awt.GLCanvas.display(GLCanvas.java:503)
        at com.jogamp.opengl.util.AWTAnimatorImpl.display(AWTAnimatorImpl.java:81)
        ... 3 more
Caused by: com.jogamp.opengl.GLException: Caught NoSuchMethodError: java.nio.FloatBuffer.rewind()Ljava/nio/FloatBuffer; on thread AWT-EventQueue-0
        at com.jogamp.opengl.GLException.newGLException(GLException.java:76)
        at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1327)
        at jogamp.opengl.GLDrawableHelper.invokeGL(GLDrawableHelper.java:1147)
        at com.jogamp.opengl.awt.GLCanvas$12.run(GLCanvas.java:1442)
        at java.awt.event.InvocationEvent.dispatch(Unknown Source)
        at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
        at java.awt.EventQueue.access$500(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue.dispatchEvent(Unknown Source)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.run(Unknown Source)
Caused by: java.lang.NoSuchMethodError: java.nio.FloatBuffer.rewind()Ljava/nio/FloatBuffer;
        at OpenGL.drawBox(OpenGL.java:410)
        at OpenGLChartsView$1.lambda$1(OpenGLChartsView.java:347)
        at java.util.ArrayList.forEach(Unknown Source)
        at java.util.Collections$SynchronizedCollection.forEach(Unknown Source)
        at OpenGLChartsView$1.display(OpenGLChartsView.java:295)
        at jogamp.opengl.GLDrawableHelper.displayImpl(GLDrawableHelper.java:692)
        at jogamp.opengl.GLDrawableHelper.display(GLDrawableHelper.java:674)
        at com.jogamp.opengl.awt.GLCanvas$11.run(GLCanvas.java:1428)
        at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1293)
        ... 16 more
Exception in thread "AWT-EventQueue-0" com.jogamp.opengl.GLException: Caught NoSuchMethodError: java.nio.FloatBuffer.rewind()Ljava/nio/FloatBuffer; on thread AWT-EventQueue-0
        at com.jogamp.opengl.GLException.newGLException(GLException.java:76)
        at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1327)
        at jogamp.opengl.GLDrawableHelper.invokeGL(GLDrawableHelper.java:1147)
        at com.jogamp.opengl.awt.GLCanvas$12.run(GLCanvas.java:1442)
        at com.jogamp.opengl.Threading.invoke(Threading.java:223)
        at com.jogamp.opengl.awt.GLCanvas.display(GLCanvas.java:503)
        at com.jogamp.opengl.awt.GLCanvas.paint(GLCanvas.java:557)
        at sun.awt.RepaintArea.paintComponent(Unknown Source)
        at sun.awt.RepaintArea.paint(Unknown Source)
        at sun.awt.windows.WComponentPeer.handleEvent(Unknown Source)
        at java.awt.Component.dispatchEventImpl(Unknown Source)
        at java.awt.Component.dispatchEvent(Unknown Source)
        at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
        at java.awt.EventQueue.access$500(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
[... 42 lines cut ...]
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue$4.run(Unknown Source)
        at java.awt.EventQueue$4.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue.dispatchEvent(Unknown Source)
[... cut ...]
        at java.util.ArrayList.forEach(Unknown Source)
        at java.util.Collections$SynchronizedCollection.forEach(Unknown Source)
        at OpenGLChartsView$1.display(OpenGLChartsView.java:295)
        at jogamp.opengl.GLDrawableHelper.displayImpl(GLDrawableHelper.java:692)
        at jogamp.opengl.GLDrawableHelper.display(GLDrawableHelper.java:674)
        at com.jogamp.opengl.awt.GLCanvas$11.run(GLCanvas.java:1428)
        at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1293)
        ... 28 more
`

danmo24, 2021-10-01:
Oh, I get it.
We need to use OpenJava and not Oracle Java.

I have installed Openjava 11 and it works fine:
https://adoptium.net/?variant=openjdk11&jvmVariant=hotspot

closed 2021-10-23 by farrellf