farrellf/TelemetryViewer #57: Why many "Lost sync with the telemetry packet stream" when connection data rate is high

liuming15, 2021-11-08:
Hi Farrell, I found your "Telemetry Viewer" as GUI software on my project. It is really good stuff and help us a lot. Appreciate your sharing the software. Now I have a question that when the connection and data rate become high, there are lots of "Lost sync with the telemetry packet stream" error occur as show below. 
[image]
I connected 3 UART to the computer and data rate of the UART is 115200 Baud rate in configuration. The really data transfer rate is on each UART is about 64.4bit per second. How can I correct the error and decrease the error rate? Thanks.

farrellf, 2021-11-08:
That happens when there is data between the end of one packet and the sync word of the next packet. You're probably sending extra data by accident, or the data structure might be incorrectly defined. You can verify this with a scope on the UART TX pin.

farrellf, 2022-06-07:
Were you able to verify the data sent to the PC? I'd like to close this issue if it's resolved.

closed 2022-09-01 by farrellf