farrellf/TelemetryViewer #58: Creating executable JAR file

Brad-HES, 2022-02-24:
This may not be an issue as much as me just not understanding what I need to add to the project to have it work right.

The downloadable JAR file for v0.8 is executable and runs with no issues.

   However, when I tried to compile the source and make an executable JAR file, I can not just click on the file for it to run. I have to run it from a command prompt and add the "--illegal-access=permit" directive to the command. 

  I have openjdk ver 16 installed.  I have added the directive to the Run/Build settings so that the program runs from Eclipse without any issue.   I can create JAR files without an issue. But I can't seem to create an executable JAR in a way to include that "--illegal-access=permit" in when it runs.  I tried searching through Eclipse help, and couldn't seem to find a way to add a command line directive to the JAR either. My temporary fix is to create a .bat file with the full command. I can double click and run that.

  I don't want to step back to anything older than ver 16. And it looks like ver 17 may not support that "--illegal-access=permit" command.

Thanks,
Brad

farrellf, 2022-02-24:
Adding an "Add-Opens" line to the JAR's manifest should work around this problem. I don't know of a way to do that with Eclipse, but it's fairly easy to do manually:

Open the JAR file with a tool like 7zip. Navigate to the "META-INF" folder, then "Edit" the "MANIFEST.MF" file. Append this line to the end of the file:

`Add-Opens: java.base/java.lang java.desktop/sun.awt java.desktop/sun.java2d`

Save the file, close Notepad, then allow 7zip to update the JAR file when prompted. Now the JAR file should run without any command line flags.

[http://farrellf.com/temp/jogl_workaround.png](url)

For anyone curious about why this is needed:

The OpenGL library that I use, JOGL, has not been updated in a while. The Add-Opens line would ideally be handled by them, and when they release a new version I assume they will take care of that. In the mean time, we have to allow access to those three packages because JOGL needs access to them.

NotEnoughHenry, 2022-03-05:
Hi, I am having this exact same issue with version 0.7. Is there any chance you know of a fix for that? Is there a specific set of settings I should have when creating the executable jar file? I added that "Add-Opens" line and yet it didn't work still.

Brad-HES, 2022-03-07:
I tried adding the Add-Opens line, and it did not work for me either.

 I've added several things to the original code that require additional libraries, so I thought that the changes I made may be the cause. I didn't try it with the original source code yet. 

  I haven't followed up on this much because the .bat file method works for now.

Brad-HES, 2022-03-07:
Thanks for this suggestion. I also had to use that to get my code to compile after I made some changes. But originally with the base 0.8 code, I didn't need that.

farrellf, 2022-06-08:
I just verified that my workaround works with TelemetryViewer v0.8 and Java 16. Here's a screencap showing before and after editing the Manifest file:

http://farrellf.com/temp/jogl_java16_workaround.mp4

So I'm guessing the problems are due to some of the changes you guys might have made. I don't know what exactly would cause it, so I would recommend reverting your changes, then gradually restoring your changes bit by bit, to figure out which change triggers a problem. Unfortunately that would probably be a long and tedious process.

Regarding the window.setVisible() problem: I'm guessing some of your code is interacting with Swing code while not in the Swing EDT. It's a common mistake, and can cause lots of weird and hard to troubleshoot problems. Most of Swing is not thread-safe.

If this issue is resolved, can we close it?

closed 2022-06-24 by Brad-HES

Masresha-y, 2022-06-25:
wow it is A perfect project it works properly but I need to add google map
with adding waypoints can you add this feature