farrellf/TelemetryViewer #61: Recommended version of Java to run with version 0.8 of TelemetryViewer

strud, 2023-02-02:
Interested to know the recommended most recent stable version of Java to run this with?

strud, 2023-02-02:
This is what I get when running the command line recommended prior by farrelf:

E:\>java --illegal-access=permit -jar TelemetryViewer.jar
OpenJDK 64-Bit Server VM warning: Ignoring option --illegal-access=permit; support was removed in 17.0
2023-02-02 18:07:48.974   [HINT    ]   Start by connecting to a device or opening a file by using the buttons below.
Exception in thread "main" java.lang.ArrayIndexOutOfBoundsException: Index -1 out of bounds for length 1
        at jogamp.opengl.windows.wgl.awt.WindowsAWTWGLGraphicsConfigurationFactory.chooseGraphicsConfigurationImpl(WindowsAWTWGLGraphicsConfigurationFactory.java:171)
        at com.jogamp.nativewindow.GraphicsConfigurationFactory.chooseGraphicsConfiguration(GraphicsConfigurationFactory.java:424)
        at com.jogamp.opengl.awt.GLCanvas.chooseGraphicsConfiguration(GLCanvas.java:1513)
        at com.jogamp.opengl.awt.GLCanvas.addNotify(GLCanvas.java:609)
        at java.desktop/java.awt.Container.addNotify(Container.java:2804)
        at java.desktop/javax.swing.JComponent.addNotify(JComponent.java:4839)
        at java.desktop/java.awt.Container.addNotify(Container.java:2804)
        at java.desktop/javax.swing.JComponent.addNotify(JComponent.java:4839)
        at java.desktop/java.awt.Container.addNotify(Container.java:2804)
        at java.desktop/javax.swing.JComponent.addNotify(JComponent.java:4839)
        at java.desktop/java.awt.Container.addNotify(Container.java:2804)
        at java.desktop/javax.swing.JComponent.addNotify(JComponent.java:4839)
        at java.desktop/javax.swing.JRootPane.addNotify(JRootPane.java:729)
        at java.desktop/java.awt.Container.addNotify(Container.java:2804)
        at java.desktop/java.awt.Window.addNotify(Window.java:791)
        at java.desktop/java.awt.Frame.addNotify(Frame.java:495)
        at java.desktop/java.awt.Window.show(Window.java:1053)
        at java.desktop/java.awt.Component.show(Component.java:1728)
        at java.desktop/java.awt.Component.setVisible(Component.java:1675)
        at java.desktop/java.awt.Window.setVisible(Window.java:1036)
        at Main.main(Main.java:151)

farrellf, 2023-02-02:
That command works for Java 16, but as the JVM reported when you ran it, Java 17 no longer supports the --illegal-access flag.

If you want to use Java 17 or newer, this can be used instead:

java --add-exports=java.base/java.lang=ALL-UNNAMED --add-exports=java.desktop/sun.awt=ALL-UNNAMED --add-exports=java.desktop/sun.java2d=ALL-UNNAMED -jar TelemetryViewer.jar

This workaround is needed because the OpenGL library I use has not been updated recently, but Java has since changed the way things work. Based on some Jogamp forum posts, it looks like that library may get a new release soon, so hopefully the next version of TelemetryViewer will not require these flags.

strud, 2023-02-02:
Thanks farrelf, this worked perfectly!

Very much appreciate the quick response.

closed 2023-02-02 by strud