farrellf/TelemetryViewer #63: Slow Updating & Freezing

NotEnoughHenry, 2023-04-01:
I don't understand where my issue lies as my feed of input is normal in the console, but when updating the displays of Telemetry Viewer, it says 40fps but clearly is at best 5fps. For hardware, I use a standard ESP32. I'm running this on my laptop, which has a dedicated GPU.

To describe the issue plainly. Having an ESP32 plugged into my laptop at all slows Telemetry Viewer to a halt and runs the software at 5fps at best. To even close Telemetry Viewer, I need to unplug the ESP otherwise, it just freezes, and not even Task Manager can shut it down.

I'm curious if there's something I don't have downloaded yet (driver-wise), but all my team members who can run Telemetry Viewer just fine don't know what I could be missing. Do you have any thoughts?

P.S. Have you tried running v0.7 on Windows 11 and know if it's compatible?

farrellf, 2023-04-26:
I have not tested it on Windows 11. I also have not heard of anyone else having a similar problem, so I doubt Windows 11 is the reason for these problems. All I can think of is the usual: install all Windows updates, and install the latest graphics driver directly from the GPU manufacturer's web site (don't wait for it to appear on Windows Update.)

NotEnoughHenry, 2023-05-06:
The slow issue then I assume, is a hardware problem with my laptop. My laptop has had issues with its motherboard, so I'll assume that's it. My drivers are up to date from NVIDIA, and Windows is as on the latest version of 10.

The Windows 11 issue was actually solved. It had to do with changing the environmental variable to the latest version they had downloaded. So once someone updated to 11, the version was changed automatically without the user's permission or choice.